tosc-rs/mnemos #27: some kind of CI

hawkw, 2022-07-19:
Now that multiple people are contributing to the repo...we should really be running some kind of CI to make sure that the code compiles, at least...

jamesmunns, 2022-07-19:
Yeah, this is a very good call. I'm going to be mostly "out of office" tomorrow, should try to get to this over the weekend.

I need to find something not using actions-rs (maybe mycelium?) to set up basic build + tests + lints.

hawkw, 2022-07-20:
feel free to copy from https://github.com/hawkw/mycelium/blob/main/.github/workflows/ci.yml, but note that it's also doing a bunch of stuff that we don't currently need (running tests in QEMU, running different sets of Loom and Miri tests based on which crates changed, etc)

PR #29 (2022-07-22): Add first CI, will likely fail fmt check
Closes #27
closed 2022-07-22 by jamesmunns