tosc-rs/mnemos #38: help wanted: smoltcp integration

jamesmunns, 2022-07-31:
It would be cool to have network capabilities from driverspace, and later pushed through to userspace via the mstd interface.

To start, it would be good to have:

* A driver service that handles the "net" layer, essentially a wrapper over smoltcp, or potentially embassy-net as it is already async
* A driver service that handles the "phy" layer, something on linux for melpo that plugs into smoltcp at layer 2/3. Later this could be used for the wifi or ethernet or usb-networking-gadget kind of interface

I'm not sure what the messaging service interface needs to look like, but my first guess is:

* Register the smoltcp-based net service w/ no phys
  * Maybe register two services, one for phy and one for net
* Start each phy, have it take a smoltcp-phy service client handle. They are not registered on service discovery, they are just tasks holding smoltcp-phy handles
* Users of the network interface can get a smoltcp-net/embassy-net service handle to open sockets, send/recv, etc.
* Write a melpo demo showing ping or a hand crafted GET response or something

jamesmunns, 2022-07-31:
For future research: Maybe we just want to give tasks a capability table at spawn time. Could be based on current_task info. Probably totally out of scope for the initial impl.

whitequark, 2023-07-30:
Came here to check if smoltcp integration is planned o:3

jamesmunns, 2023-07-30:
Hey @whitequark, likely soon! @hawkw has started working on getting an ESP32C3-based wifi firmware going in #191, which will likely act as a "dumb" 802.3 interface, with smoltcp handling the actual network stack on the host.

We'll likely use (or base?) our work on embassy's `embassy-net` crate, as it already provides an async wrapper for smoltcp.

whitequark, 2023-07-30:
Nice!