tosc-rs/mnemos #49: Tracking Issue: Start of June cleanup

jamesmunns, 2023-06-01:
CC @hawkw, there are a couple things in flight across a couple of repos. It might be good to pause for a bit and do a little cleanup before moving forward some more. From my perspective:

* [ ] We want to get `mycelium` dependencies updated
    * [x] https://github.com/tosc-rs/mnemos-alloc/pull/7
    * [ ] https://github.com/tosc-rs/mnemos-alloc/pull/9
    * [ ] https://github.com/tosc-rs/mnemos/issues/47
* [ ] We (probably) want to merge the following repos into `mnemos` as a monorepo:
    * [ ] https://github.com/tosc-rs/mnemos-alloc
    * [ ] https://github.com/jamesmunns/forth3/
        * There are a couple semi-related projects in here. Not sure what to do about `forthboy` and `f3repl`, but they could find a new home
    * [ ] https://github.com/tosc-rs/teletype/
* [ ] There are some features in-flight right now (*mostly* in forth3)
    * "Bag of Holding"
        * [ ] https://github.com/jamesmunns/forth3/pull/19
        * [ ] https://github.com/tosc-rs/mnemos/pull/48
    * [ ] https://github.com/jamesmunns/forth3/pull/18
    * [ ] https://github.com/jamesmunns/forth3/pull/17
    * [ ] https://github.com/jamesmunns/forth3/pull/5
    * [ ] https://github.com/jamesmunns/forth3/pull/2
* [ ] Clean up clippy lints across all projects

I'll stop adding more features for now, at least until we can get deps updated and move to a monorepo. I think my open features can wait for now, and I can rebase their changes onto the new monorepo.

For the monorepo merge, I'll probably plan to do an "unrelated history merge" (and NOT squash the merge), to retain the history of the side projects.

jamesmunns, 2023-06-01:
Also my $0.02, I think we should:

* Merge the mycelium deps update
* Merge the bag of holding changes
* Merge the `drop_dict` change
* Then do the monorepo merge
* Then "recover" the unmerged forth3 prs

hawkw, 2023-06-04:
FWIW, I'm honestly less attached to whether or not forth3 merges, as it does feel weird to have the other little projects in tree when they're not technically part of mnemos...I think having the allocator in tree is important though.

jamesmunns, 2023-06-04:
So, I think we hit the important things here! I'll go ahead and close this, thanks for all the help @hawkw!

closed 2023-06-04 by jamesmunns