tosc-rs/mnemos #71: Tracking Issue: Beepberry Computer v0.1

jamesmunns, 2023-06-13:
CC @hawkw, this is just a running list for "it's a computer" feature completeness on the beepberry, in no particular order

* [ ] Restore the the SPI driver from `james/add-sharp-display` (I forgot to grab this in #69)
* [ ] Move the `EmbGraphics` driver definition from melpo to the kernel
* [ ] Implement the `EmbGraphics` driver on top of the SPI port for `lichee-rv`
* [ ] Write an I2C driver for the D1, and create some kind of service interface for it
* [ ] Integrate https://github.com/jamesmunns/bbq10kbd, on top of the driver service
    * [ ] Figure out if there are any changes necessary for sqfmi's fork of the keyboard firmware
* [ ] Make a new `platform/allwinner-d1/boards/mq-pro` project, make sure everything from `lichee-rv` still works there.

I *think* these are all the "must dos" to have a standalone computer. From there, we can either start cleaning up some "nice to haves", or start figuring out what to build next

jamesmunns, 2023-06-20:
Did a quick cross check of pinouts on the Pi/lichee-rv/mangopi mq-pro [on a spreadsheet](https://docs.google.com/spreadsheets/d/1XbbXARVXu009ZRdzl69T0_YmjL-_Ab0TnPQSrKdww2I/edit?usp=sharing).

Looks like the same SPI pins are used for the display, so no changes to the code in #72 should be necessary.

UART0 is also brought out to the same pins as the standard RPI UART TX/RX:

[image]

jamesmunns, 2023-06-20:
Unfortunately the I2C lines available on the MQ-Pro (PG12/PG13) are not brought out on the Sipeed board (PB0/PB1 are used instead).

PG12/13 are mapped to TWI0 (and also UART1).
PB0/1 are mapped to TWI2 (and also UART2).

Interestingly enough, it looks like we could remap the I2C to UART, the RP2040 has a matching UART pair on the same pins as SDA. This could be useful for higher bandwidth comms.

jamesmunns, 2023-06-20:
Also of note: The Mango Pi PTH are too small to work with the pogo pins on the beepberry. For now, I have designed a small adapter board that will give us space to mount the mango pi, and also additional breakout pins for testing. It should arrive around 2023-06-23 or so for testing.

[image]

In the long term, it might be better to design our own board based on a sipeed lichee-rv SOM.

I found [this project](https://github.com/riktw/jeadock) which has already done this to breakout ethernet capabilities, we can likely re-use their pinout/sizing.

Making a new board will also let us do a couple things that are interesting:

* Using a different (less of a pain in the ass) wifi chip, like an esp32c3
* Breakout the onboard headphone/mic audio I/O
* Use other IO available (RPI header is limited to 40 pins, sipeed SOM has 134 pins, both are a mix of I/O and power)

hawkw, 2023-06-20:
@jamesmunns are you currently working on the I2C driver? If not, I'd be happy to pick it up.

jamesmunns, 2023-06-20:
@hawkw not yet! Very happy to have you work on it. Let me know if you need/want help with anything.

hawkw, 2023-07-16:
as of #155 and #158, the keyboard works and is glued onto the Forth shell, so i think we can call this one finished! 🎉🎉🎉

closed 2023-07-16 by hawkw