tosc-rs/mnemos #74: Design Topics around services + messaging

jamesmunns, 2023-06-19:
So here are a couple things I am realizing as I am working on services and stuff. This should probably be a couple issues/action items, but I am dumping my thoughts here instead of in chat.

## We should think about our use of `alloc` in kernel services

Originally, I loosely had the plan that "services can alloc when starting, but shouldn't alloc later". This is why you see all the "how many clients/messages/frames does this service take" in `register()` calls.

On one hand: it'd probably be good for smaller systems to have bounded memory usage, more or less just acting like a bump allocator on startup, and not requiring variable runtime allocation.

On the other hand: It's sometimes a pain in the butt to actually do this/very convenient to alloc whenever.

I dunno if there is a "right" answer, but picking one and being consistent is probably a good thing.

## Messaging should be easier/better encapsulated

There's a LOT of impl details exposed to every service. The core interface is:

* Servers receive messages (from many clients)
* Servers optionally reply to messages

There is SOME shenanigans whether the message types are `Serialize` or not, that limits whether a server can advertise ONLY to other kernel client entities, or (in the future) userspace or proxied/remote client entities, but right now there are a ton of `Message`/`Envelope` whatever whatever types that are kinda gross to actually use.

This could probably use some better encapsulation, and nicer APIs on both the server and client side.

## Service definition should be more consistent

I'm sort of learning how services "should be" as we are writing them.

I think at the moment, my best practice guide is:

* Services should be defined (whenever possible) in the kernel
* Services should provide a reasonable client implementation
* `RegisteredDriver` should be implemented on a "nothing" type, that can be used to get a client `Nothing::from_registry() -> NothingClient` or so.
* Services defined in the kernel SHOULD NOT provide a Server implementation, unless they only rely on other services (like `SerMux` does).

hawkw, 2023-06-19:
> ## We should think about our use of `alloc` in kernel services

Another "lesson learned" for me is that, with the async `mnemos-alloc` API, the failure mode for OOM conditions is "a task trying to allocate memory waits forever". When some allocations are dynamic and short-lived, this isn't terrible, as that memory may later be released and the task can just wait until its available. However, if the allocation occurs while *initializing* a kernel task that lives for the entire runtime of the OS, such as a driver service, and the only other existing allocations are for similarly long-running driver services, the whole system just gets completely stuck. And, because the allocator doesn't have any logging or similar diagnostics, there's no way to determine *why* the system is stuck.

It might be advantageous to add a mode switch to the allocator which makes allocations panic if there's no available memory, rather than wait for it to become available. We could have the kernel switch modes so that during initialization, if there's not enough heap space to initialize all the driver services, the system panics so that the kernel developer knows we can't run all the necessary kernel driver services in the current heap configuration. Once we've finished initializing driver services and are running user tasks, we would switch to the current `mnemos-alloc` behavior and always wait on OOM conditions.

Alternatively, we could just add a fallible `try_alloc_FOO` family of methods to the heap, and have driver services prefer to use those over the asynchronous `alloc_foo` methods during their initialization.

jamesmunns, 2023-06-19:
I think *if we stick with the async allocator*, it would make sense to do one or both of:

* Add a way to "block on" the `initialize` items to complete, or reach some level of readiness (todo: how to determine this?)
* Add a flag to force the allocator into "now or never" mode, where OOM panics, until we are done with the initialization phase.

Reading a little bit about how the [linux kernel handles allocation](https://www.kernel.org/doc/html/v5.0/core-api/memory-allocation.html), it might make sense to have a couple of allocation features anyway, especially if/when we ever get to caching and paging of memory, where having something async (like resident memory wired) might make more sense to have async options, in addition to immediate mallocs.