tosc-rs/mnemos #108: Consider AsyncRead/AsyncWrite traits

jamesmunns, 2023-06-27:
So in https://github.com/tosc-rs/mnemos/blob/e8b17b6db412b90b1e4e5e4e431c5f30b1926b3e/source/kernel/src/forth/shells.rs#L22-L47, I was able to get a forth shell over sermux working pretty darn easy, but we need to have a little glue task that forwards messages between the two pipes.

I did look briefly at "extracting" the pipes, but realized that `PortHandle` does the crowtty encoding in the client, which means you can't "just" give the underlying bidi to forth via the "new_with_stdio" method introduced in #106.

If we had async read/write traits, we might be able to do this more easily. I don't think this is *exactly* a problem today, but I can see this "shape" of problem occuring again in the future.

jamesmunns, 2023-06-27:
https://github.com/tosc-rs/mnemos/assets/2097964/c63b5c19-fafb-4ce7-b39a-00c282a4cc2d

hawkw, 2023-06-27:
I think our `AsyncRead`/`AsyncWrite` traits will probably need to look pretty different from the ones in both `futures` and `tokio`, because of how our kernel services work. The `tokio` and `futures` `AsyncRead`/`AsyncWrite` traits work by passing in borrowed/mutably borrowed slices as read/write buffers. Since our drivers are iimplemented as kernel services that we talk to over a channel, we can't easily hand them borrowed slices. Instead, I think we want `AsyncRead`/`AsyncWrite` traits that take ownership over a buffer, and then return the buffer to the caller when the read/write is complete, so that the caller can reuse it.

I'm imagining something like this:
```rust
pub trait AsyncRead {
    /// Read up to `len` bytes into `buf`, returning the buffer.
    async fn read(&mut self, mut buf: OwnedReadBuf, len: usize) -> Result<OwnedReadBuf, io::Error<OwnedReadBuf>>;
}

pub trait AsyncWrite {
   /// Write up to `len` bytes from `buf`, returning the buffer.
   async fn read(&mut self, mut buf: OwnedWriteBuf, len: usize) -> Result<OwnedWriteBuf, io::Error<OwnedWriteBuf>>;
}
```

I started sketching an `OwnedReadBuf` type in [this branch](https://github.com/tosc-rs/mnemos/compare/eliza/readbuf?expand=1), as part of the I2C stuff. I can try to expand on that to introduce read/write abstractions if you like?