tosc-rs/mnemos #120: Tracking Issue: Beepberry Hardware Support v0.2

jamesmunns, 2023-07-02:
This is a follow-up for things that would be good to support once all the items in #71 are complete.

## Booting from SD Card (or SPI flash)

This is probably going to involve a couple things:

* Figuring out how to prepare a binary image that the D1 will recognize and start booting from
* Writing some kind of early stage bootloader (could be the whole kernel at first lol)
* Figuring out how much of the SD card the hardware will load for us
* Initializing the DDR3 memory
* Booting without the help of XFEL

I think the oreboot/rustsbi crates have at least figured out part of this (preparing an image, and initializing the DDR3 memory), though it may have been from a SPI flash and not an SD card, though the stock MQ-Pro boards do not come with SPI flash, so we would have to solder our own on.

I've been looking around, and haven't found any docs on what the SD card image is supposed to look like, but there is now apparently support in U-Boot for whatever this format is.

## General SD Card Support

For general operation, we probably will want to be able to read data from the SD card, and then eventually be able to write to it. We may need this in the early stage bootloader, depending on how much the hardware does for us on boot.

For convenience, this probably also means porting one of the FAT fs libraries as well, on top of raw block operations.

## USB OTG support

The details of this are tracked in #88. We will want this for "better than UART" I/O, especially until some kind of networking is supported

## ESP32C3 wifi

Supporting the RTL wifi chip is likely... infeasible. We should look into using something like esp-hosted, or writing our own firmware for the ESP32 to act as a wifi modem over SPI.

This will likely also entail supporting smoltcp (#38) as a network stack.

hawkw, 2023-07-02:
we could probably put a spi flash for our bootloader on the adapter board, along with an esp32c3, i imagine?

jamesmunns, 2023-07-02:
Oh. Yeah. That would be a totally straightforward thing to do.

hawkw, 2023-07-02:
(i'm getting the feeling that basically every problem can be solved by forcing James to spend more time in KiCad... :D)

jamesmunns, 2023-07-02:
I asked @orangecms about oreboot/rustsbi:

> Hey Daniel aka CyReVolt 🐢 - I think you said you had gotten DRAM init and flash loading working for the D1 for rustsbi or oreboot, am I remembering that correctly?

He responded:

> Yes, both - we do DRAM init in oreboot, and then use RustSBI (a library crate) to implement SBI, can boot into Linux. :)
>
> https://github.com/oreboot/oreboot/tree/main/src/mainboard/sunxi/nezha#allwinner-nezha--d1
>
> it works for 512MB RAM, with 1GB it gets a bit finicky, needs some fixing. 😬
smaeul's U-Boot fork has the reference code in C that Paul Ruizendaal reversed from the asm dump that Allwinner (sunxi) provided:
>
> https://github.com/smaeul/u-boot/commit/1fd4bcd7b40bcf5e3bd757dc6b7b4dbb3cfef52b

orangecms, 2023-07-02:
All that the D1 needs to boot from any media is the special eGON header. On SD cards, it needs to be at a specific offset. See here further down: https://linux-sunxi.org/Allwinner_Nezha#U-Boot

We have the header implemented in oreboot; see the sunxi/nezha mainboard directory, ant bt0/main.rs under it. A checksum is filled in via xtask.

jamesmunns, 2023-07-02:
Ah ha! The "eGON header" was the thing I couldn't find the name of. [this wiki link](https://linux-sunxi.org/EGON) explains the format.

This helps a ton, thank you :)

jamesmunns, 2023-07-26:
Opened #191 to specifically track the "ESP32C3 wifi" part of this.