tosc-rs/mnemos #121: Idea Collector: Beepberry Shield

jamesmunns, 2023-07-02:
So, we've had some ideas to make a mnemos shield since the MQ-Pro doesn't check as many boxes as we'd originally hoped. Collecting possible options here:

* Use a Lichee-RV SOM, so have 2x m.2 sockets
    * Has SD-Card and USB-OTG port on the SOM
* Have a QSPI flash for bootloader and for persistent image before we have SD Card support
* Have an ESP32C3 for wifi
* Have a lot of breakout pins for development

There are a ton of other options that I don't think would be immediately useful and would add complexity to the board, but maybe @hawkw disagrees:

* Ethernet PHY/port
* USB-A host port
* Some kind of higher resolution display (mipi, hdmi) option

hawkw, 2023-07-02:
I agree that we probably don't want Ethernet or HDMI on the board; an ethernet jack is gonna make the whole stack even taller, which feels kinda sad, and the MQ Pro already has mini-HDMI on the board...

hawkw, 2023-07-02:
A Lichee RV-based breakout does seem like it could potentially stick out a bit less far than a MQ Pro and header/socket...I would assume the dual-M.2 thingy lets us access more pins from the SOC on the board than just the 40-pin Pi GPIO header?

jamesmunns, 2023-07-02:
Yeah, I have to check the pinouts, but there's a ton more on the lichee-rv. It's like 2x67 pins total, not sure how many are power only

jamesmunns, 2023-07-02:
[image]

hawkw, 2023-07-08:
@jamesmunns suggested on matrix that we just put XIAO/QT Py headers on the breakout for mounting a C3 dev board, instead of actually adding a SMD esp32c3:
> oh! I could put a XIAO pad on it for the C3!
> that would actually be pretty nice, and reasonably sized, and very removable for testing
> that would actually be pretty nice, and reasonably sized, and very removable for testing

hawkw, 2023-07-08:
also, future revisions of the PCB should include a switch or solder pads or something on the 5v rail — right now we've had to do PCB surgery to disable it so we can use USB for flashing.

hawkw, 2024-04-26:
It would be cool if a future revision broke out the Pi header's UART pins into a FTDI or similar USB-serial guy, so that there's no need for an external dongle hanging off the side with dupont wires.

hawkw, 2024-04-26:
We could probably cross off the QSPI flash since we do have SD boot working already. But, it could still be cool to have.