tosc-rs/mnemos #129: design: perf counters, metrics, things of that nature

hawkw, 2023-07-05:
we should probably have them!

- i wrote a little prometheus library (https://github.com/hawkw/tinymetrics/) but i dunno if we actually want prometheus here (text format...)
- how do metrics get read from? do we stick even more stuff on our measly 115200-baud serial port?
- host should probably pull metrics (rather than target pushing metrics) so you don't waste a bunch of time serializing metrics that nobody wants.
- what metrics are nice to have?

hawkw, 2023-07-05:
probably a good steelthread use case is "how much heap have we used?"

because it would be really nice to have some sense of the space our pile of tasks already takes up, and to know if we're hanging because we're OOM, or some other thing.

hawkw, 2023-07-05:
on the other hand, the disadvantage of pull metrics is that we have to ask the target to give us metrics, and if it's totally deceased, we have no way of looking at its metrics from before it went casters-up (unless we had previously asked for them). 

but, we basically only go totally dead for a few reasons:
- infinite loop
- interrupt that doesn't get cleared (this is just an infinite loop but with more steps)
- panic
- scheduler's haunted

panics are probably a relative non-issue because we can just make the panic handler scream every counter we've ever recorded into the void prior to dying.

jamesmunns, 2023-07-06:
So in terms of memory usage, the LLA crate can give us a "used/free" number on a pull basis, we could also keep some atomic counters in the allocator as well, that subtract and add on alloc/free. If we wanted to make it push instead of pull, we could do this once per scheduler tick (or some interval of that), or the allocator adapter could hit a static wakecell on every allocation or free (this might be too much, open to other ideas).

If we end up making some way of "get current task" from a global scope, the allocator adapter could also potentially update metadata in the task "header", such as tracking allocations per task, as well as allocations made outside of a task context, such as in calls to `initialize()` or the kernel event loop itself. The downside of this is that things like "allocate a buffer and pass it by ownership to a task" would be allocated in the parent, and freed in the child, so we might just want to have separate "amount allocated" and "amount freed" counters rather than one allocation `usize`.

hawkw, 2023-07-06:
my current thinking is that we should have some kind of global collection of metrics that can be pulled either by crowtty (maybe adding new messages to the trace protocol?) or by applications running on the device (e.g. eventually we could have a "metrics" program that you can run to display the current metrics). when crowtty is connected, we would proactively pull metrics snapshots occasionally, in case the target goes into a bad state. we would also include a metrics snapshot in our panic message. eventually, when we figure out a nicer boot situation than "using xfel to overwrite DDR every time", we would probably want to dump any metrics from a previous run when coming back after a crash, as well. 

this way, we're not constantly spending time pushing metrics when nobody is listening to them, but we can have crowtty proactively pull in case of a crash.

hawkw, 2023-07-16:
error counters for different components would probly be good

hawkw, 2023-09-15:
#269 adds code to collect some basic stats in `mnemos-alloc`. Now we need to actually add a way of exposing that to the user...