tosc-rs/mnemos #146: "Needs Docs" Job Queue

jamesmunns, 2023-07-11:
Please leave comments here for things that need docs, and we'll document them!

spookyvision, 2023-07-11:
My questions were already all answered but it could probably use a writeup:

how to use `kernel.tick()` and `timer.{force_}advance` to build a main loop

hawkw, 2023-07-11:
- [ ] it would also probably be nice to have all the dev environment setup docs in centralized location (e.g. maybe a `BUILD.md` or `CONTRIBUTING.md` --- we have a bit of stuff [in the README](https://github.com/tosc-rs/mnemos/blob/main/README.md#getting-started), but not very much, and the D1 platform impl has [this](https://github.com/tosc-rs/mnemos/blob/main/platforms/allwinner-d1/README.md#flashing-a-d1-sbc) in its own README, but i'm not sure how easy it is to find any of that. there's also no "here's where to install `just` so you can use the `justfile`" instructions
- [ ] crowtty is completely undocumented except for its CLI help

hawkw, 2023-07-11:
"how to make a kernel event loop" docs can probably reference https://mycelium.elizas.website/maitake/scheduler/index.html#executing-tasks and https://mycelium.elizas.website/maitake/time/timer/struct.timer#driving-timers

Aloso, 2023-07-11:
Please set the `git-repository-url` option (documented [here](https://rust-lang.github.io/mdBook/format/configuration/renderers.html#html-renderer-options)), then mdBook includes a GitHub link in the top right corner on every page 🙂

hawkw, 2023-07-14:
- [ ] pomelo should probably be mentioned in the main readme