tosc-rs/mnemos #159: platform(d1): nice abstraction for a task to get woken by a GPIO interrupt

hawkw, 2023-07-15:
so we can stop busy-polling the keyboard. it would be nice if the i2c_puppet driver could be still cross-platform and just depend on the cross-platform I2C driver and some kind of GPIO IRQ interface provided by the actual board impls, but idk what that would look like...

hawkw, 2023-07-15:
@jamesmunns curious if you have any thoughts on this. i might try to take a crack at it soonish

jamesmunns, 2023-07-15:
@hawkw no specific feedback, other than it would probably be good to follow Embassy's lead on APIs and such. For example, https://docs.embassy.dev/embassy-rp/git/rp2040/gpio/struct.Input.html

We don't need to impl all the methods, but in general a "wait for X edge" is good.