tosc-rs/mnemos #160: Idea collector: Configuration System

jamesmunns, 2023-07-15:
We currently have a lot of magic numbers for things like buffer sizes, number of ports, etc. We also don't particularly provide any guidance for what these numbers should be. In the immediate future, it would be good to have one central "config" table that can be used across the system for setting (and viewing the settings of) these values, even if the values are "baked in" at compile time.

In the future, we might want to have other neat features, such as:

* Layering of configs, so you can say "gimme all the defaults BUT change/add these"
* Loading from a file from disk at boot, instead of only at compile time
* Potentially changing these settings at runtime, if we ever do/want to support hot-reloading of services, so you can edit a config file, and restart a service

I think feature wise we'll probably also want some kind of namespacing, so we can get specific config values, like "sermux outgoing frame size".

I think there's a few main stylistic ways to go about this:

* Go all strings, using a very simple key:value system, where all settings are basically `&[(&str, &str)]`. Probably the simplest to write, pushes all the parsing + validation down to the consumers of the settings. We could potentially take a JSON or TOML at compile time and convert it to the "settings array" with a build-rs or something.
    * example for sermux: `("services/54c983fa-736f-4223-b90d-c4360a308647/max_frame", "512")`
* Keep simple string keys + array of tuples, but also support a limited number of primitive "value" types, such as `u8`, `i32`, `usize`, `str`, etc. Could probably be used mostly the same, but gives a little more type safety, but still there is a potential for "we wanted a `u8` but got an `i32`, what do?".
    * example for sermux: `("services/54c983fa-736f-4223-b90d-c4360a308647/max_frame", CfgVal::usize(512))`
* Allow for some kind of namespace schema, possibly not for ALL namespaces, but for some. This might require adding a `Settings` parameter to the `RegisteredService` trait, which we've talked about informally before. I'm not sure how to do this best, but we could likely rely on serde (with JSON/TOML/YAML/Postcard backends) for decoding.
    *  example for sermux: `cfg.add_service_cfg<SerialMuxService>(SerialMuxSettings { max_ports: 16, max_frame: 512 });`
    * Open question: how would this work for NON service items, like user apps? Do they have to "fall back" to stringly typed data? Is there some non-service way of adding configs?
    * How to handle "array of Any types?", something like BoH? Keep it serialized (as a `Vec<u8>`) and deser on demand?
* Some other way? something like `kconfig`? Something with a lot of cursed proc macros like `clap` with complicated validation capabilities?

Whatever this is, it probably needs to exist at the kernel and/or global level, rather than just as part of the registry, as users will likely want to use the same single config system for app and daemon configuration as well.

There might also be fun design challenges around runtime-loading: we might require an `fs` service to be running to load the config from disk, but then how can we configure the fs service? At the moment the kernel has no "service dependencies", the only required capabilities are the allocator and the timers running (before first tick), and getting configuration for (at least) all other services might require (for example) launching an SDIO service (at the hardware level), then an fs service that sits on top of that. This doesn't need to be solved for v1 though.

hawkw, 2023-07-15:
I'm strongly +1 on the need for something like this, it would be really nice not to have ad-hoc config values scattered all over the place. 

This could be a good project for someone who's interested in getting more involved in mnemOS development, since it:
- touches a lot of different parts of the project, and exposes you to lots of different components
- is a decently large chunk of work
- doesn't require a really deep knowledge of our hardware targets
- I really want to have it but I also have a bunch of other stuff in my queue right now (and i imagine @jamesmunns does too)

hawkw, 2023-07-15:
In any case, this feels like a big enough piece of design work that it might be smart to start off with an (informal) RFC before implementing it, so that the design is documented up front.

jamesmunns, 2023-07-15:
> In any case, this feels like a big enough piece of design work that it might be smart to start off with an (informal) RFC before implementing it, so that the design is documented up front.

This is kind of the pre-rfc for that :D

hawkw, 2023-07-15:
> This is kind of the pre-rfc for that :D

yup, i figured! :) just thought it would be good to make sure the final design gets written down.

hawkw, 2023-07-15:
i wonder if @jamesmunns' `toml-config` crate would be a good fit for this...

hawkw, 2023-07-15:
another thought (whoops, now i'm a bit nerdsniped)...we could probably add something to the `RegisteredDriver` trait to define its config key, and perhaps also its confit _type_. something like:

```rust
pub trait RegisteredDriver {
    const CONFIG_KEY: &'static str;
    type Config: serde::Deserialize + Default;
    // rest of the trait goes here
}
```
then, you could just have an API like
```rust
impl RegisteredDriver for SerialMuxService {
    type Config = SerialMuxSettings;
    // ...
}

kernel.get_config::<SerialMuxService>() // returns SerialMuxSettings
```

i guess this wouldn't even *need* to be part of `RegisteredDriver`, it could be a whole separate trait, too, if that made more sense.

PR #199 (2023-07-30): RFC: Basic Configuration
This RFC introduces a high level plan for implementing a configuration system, as discussed in #160.

Closes #160
closed 2023-07-30 by jamesmunns