tosc-rs/mnemos #170: on-device testing

hawkw, 2023-07-19:
it would be cool to be able to run automated tests on a target board using a test runner tool on the host. we could drive the tests over SerMux. 

if the tests exercise functionality that can currently be accessed from Forth, this could be as simple as just opening a serial forth shell, sending a script, and making assertions about the output.

hawkw, 2024-04-14:
It would be really neat to get this working. If we could come up with a way to do hardware-in-the-loop testing via Crowtty, the next step could be a way to do automated hardware-in-the-loop testing for CI (I'm imagining a GitHub Actions self-hosted runner on a Raspberry Pi or something hooked up to a D1).

hawkw, 2024-04-14:
> automated hardware-in-the-loop testing for CI (I'm imagining a GitHub Actions self-hosted runner on a Raspberry Pi or something hooked up to a D1).

ahahahha i just realized that this would currently be a huge pain in the ass to get working for an _incredibly stupid_ reason: an `xfel`-based flashing workflow can't flash the D1 again until it's been power-cycled, which we currently do by...physically unplugging the OTG USB port. We'd need a way for our test-driver box to softwarily reset the device. there's probably some obscure `ioctl` or something that a linux machine could use to stop powering one of its USB ports?

hawkw, 2024-04-14:
> there's probably some obscure `ioctl` or something that a linux machine could use to stop powering one of its USB ports?

ah, [you can write `on` or `off` to `/sys/bus/usb/devices/<BUS ADDR>/power/control`](https://www.baeldung.com/linux/control-usb-power-supply) because Everything Is A File. that's actually really nice!

spookyvision, 2024-04-16:
for completeness' sake, the command sequence seems to be:
```sh
# turn off (suspend)
$ echo 0 > power/autosuspend_delay_ms
$ echo auto > power/control

# device should now autosuspend immediately
$ cat power/runtime_status
suspended

# turn back on
$ echo on > power/control

# kernel should now emit messages about resetting, disconnecting and rediscovering the device
# (...)

$ cat power/runtime_status
active
```

spookyvision, 2024-08-13:
an additional unfortunate snag is that once we boot mnemOS we lose XFEL mode. In other words, for HIL we'd need to add actual power down, "push" XFEL and power up again :(