tosc-rs/mnemos #171: platform(melpomene): simulate I2C devices

hawkw, 2023-07-19:
it would be cool if we could exercise the `i2c_puppet` driver in Melpomene. in order to simulate i2c devices, we would need to add a simulated implementation of the `I2cService` in Melpomene, and configure it to respond to I2C transactions with the `i2c_puppet` address.

spookyvision, 2023-07-21:
copied from chat:

(Eliza)
> specifically what i'm thinking would be nice for melpomelo is if we could have a sim impl of I2cService so that the i2c_puppet driver, which is cross-platform, can run in melpo
> it would be really cool if melpo made keyboard input pretend to be i2c_puppet and then the i2c_puppet driver read from the IRL keyboard

(me)
> Okay, so just the service for starters
> I was figuring you'd want some kind of instrumentation, like "expect this sequence of commands"

spookyvision, 2023-07-21:
related: #169