tosc-rs/mnemos #225: bug(just): Just recipes don't gate x86_64 correctly on aarch64 hosts

jamesmunns, 2023-08-10:
Running `just check` or `just clippy` fail on x86_64 due to missing imports and inline asm:

```
cargo clippy --lib --bins --examples --tests --benches --all-features -- -Dwarnings
    Checking sermux-proto v0.1.0 (/Users/james/personal/mnemos/source/sermux-proto)
   Compiling mnemos v0.1.0 (/Users/james/personal/mnemos/source/kernel)
    Checking hal-x86_64 v0.1.0 (https://github.com/hawkw/mycelium#1f125194)
    Checking crowtty v0.1.0 (/Users/james/personal/mnemos/tools/crowtty)
error[E0432]: unresolved import `core::arch::x86_64`
 --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu/intrinsics.rs:3:21
  |
3 | pub use core::arch::x86_64::{
  |                     ^^^^^^ could not find `x86_64` in `arch`

error: invalid register `ecx`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu/msr.rs:278:17
    |
278 |                 in("ecx") self.num,
    |                 ^^^^^^^^^^^^^^^^^^

error: invalid register `eax`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu/msr.rs:279:17
    |
279 |                 out("eax") lo,
    |                 ^^^^^^^^^^^^^

error: invalid register `edx`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu/msr.rs:280:17
    |
280 |                 out("edx") hi,
    |                 ^^^^^^^^^^^^^

error: invalid register `ecx`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu/msr.rs:301:13
    |
301 |             in("ecx") self.num,
    |             ^^^^^^^^^^^^^^^^^^

error: invalid register `eax`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu/msr.rs:302:13
    |
302 |             in("eax") lo,
    |             ^^^^^^^^^^^^

error: invalid register `edx`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu/msr.rs:303:13
    |
303 |             in("edx") hi,
    |             ^^^^^^^^^^^^

error: invalid register `dx`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu.rs:119:27
    |
119 |         asm!("in al, dx", in("dx") self.num, out("al") result);
    |                           ^^^^^^^^^^^^^^^^^

error: invalid register `al`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu.rs:119:46
    |
119 |         asm!("in al, dx", in("dx") self.num, out("al") result);
    |                                              ^^^^^^^^^^^^^^^^

error: invalid register `dx`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu.rs:127:28
    |
127 |         asm!("out dx, al", in("dx") self.num, in("al") value)
    |                            ^^^^^^^^^^^^^^^^^

error: invalid register `al`: unknown register
   --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/cpu.rs:127:47
[... 160 lines cut ...]
    | |_________________^
...
560 | /         gen_code_faults! {
561 | |             self, H,
562 | |             Self::DIVIDE_BY_ZERO => fn div_0_isr("Divide-By-Zero (0x0)"),
563 | |             Self::OVERFLOW => fn overflow_isr("Overflow (0x4)"),
...   |
569 | |             Self::X87_FPU_EXCEPTION => fn x87_exn_isr("x87 Floating-Point Exception (0x10)"),
570 | |         }
    | |_________- in this macro invocation
    |
    = note: this error originates in the macro `gen_code_faults` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0599]: no function or associated item named `new` found for struct `CpuId` in the current scope
 --> /Users/james/.cargo/git/checkouts/mycelium-fb863cdaca369b67/1f12519/hal-x86_64/src/time/tsc.rs:9:16
  |
9 |         CpuId::new()
  |                ^^^ function or associated item not found in `CpuId`

Some errors have detailed explanations: E0432, E0570, E0599.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `hal-x86_64` (lib) due to 39 previous errors
warning: build failed, waiting for other jobs to finish...
error: Recipe `clippy` failed on line 64 with exit code 101
```

hawkw, 2023-08-10:
we probably need `forced-target = "x86_64-unknown-none"` in the `mnemos-x86_64-core` crate's Cargo.toml, if it isn't already.

hawkw, 2023-08-11:
Ugh, unfortunately, it seems like any crate that has a `forced-target` cannot currently be in the workspace's `default-members` (which are checked/built with a bare `cargo check` without a `-p`), since they result in core/alloc not being present when checking the whole workspace. Because we can't solve this using build-std (see https://github.com/rust-lang/cargo/issues/9451 and https://github.com/rust-lang/cargo/issues/8733), it seems like we probably need to remove the x86_64 target from `default-members` for now...

PR #226 (2023-08-12): chore: add `forced-target` for `x86_64-core`
This should result in it no longer breaking the build on non-x86_64
build host platforms. Unfortunately, it also means we need to remove the
 `mnemos-x86_64-core` crate from the workspace's `default-members`. See:
 https://github.com/tosc-rs/mnemos/issues/225#issuecomment-1675471449

Fixes #225 (I hope; @jamesmunns, can you check this on your M1 mac?)
commit 455050b3 (2023-08-12): chore: add `forced-target` for `x86_64-core` (#226)

This should result in it no longer breaking the build on non-x86_64
build host platforms. Unfortunately, it also means we need to remove the
 `mnemos-x86_64-core` crate from the workspace's `default-members`. See:
 https://github.com/tosc-rs/mnemos/issues/225#issuecomment-1675471449

Fixes #225

---------
closed 2023-08-12 by jamesmunns