tosc-rs/mnemos #265: CI build for pomelo

spookyvision, 2023-08-31:
came up with this GHA config, verified it works locally with `act` (when run from the pomelo subdir, with the configuration placed in `platforms/pomelo/.github/workflows/`). When using the `actions-rs/toolchain@v1` action for real, we should remove the `toolchain` variable - according to the [documentation](https://github.com/actions-rs/toolchain/blob/master/action.yml#L14C57-L14C57) it then picks up what's specified in `rust-toolchain.toml`.

```yml
on: [push, pull_request]

name: Pomelo

env:
  TRUNK_VERSION: v0.17.5

jobs:
  build:
    name: Build
    runs-on: ubuntu-latest
    strategy:
      matrix:
        target:
          - wasm32-unknown-unknown
    steps:
      - uses: actions/checkout@v2
      - uses: actions-rs/toolchain@v1
        with:
          toolchain: nightly
          target: ${{ matrix.target }}
          override: true
      - name: trunk build
        run: wget -qO- https://github.com/thedodd/trunk/releases/download/${TRUNK_VERSION}/trunk-x86_64-unknown-linux-gnu.tar.gz | tar -xzf - && ./trunk build --release --public-url="/misc/pomelo"
```

spookyvision, 2023-08-31:
James:
> We can also use cargo-binstall which also has an action we use for nextest and stuff: https://github.com/cargo-bins/cargo-binstall#faq
> e.g. https://github.com/tosc-rs/mnemos/blob/main/.github/workflows/ci.yml#L86-L87

hawkw, 2023-09-04:
@spookyvision  do you wanna open a PR to add this?

hawkw, 2023-09-04:
Ah, got it. Never mind!