tosc-rs/mnemos #285: `cargo mn` throws some warns on first build

jamesmunns, 2023-09-18:
Tried running `cargo mn` (with no args) today, and it printed some warnings we might want to address.

```
$ cargo mn
warning: skipping duplicate package `run-wasm` found at `/Users/james/.cargo/git/checkouts/embedded-graphics-web-simulator-ef34bb7f9f200182/0f07b1e/examples/animated/run-wasm`
  Downloaded cargo-util v0.2.5
...
    Finished release [optimized + debuginfo] target(s) in 1m 25s
     Running `target/release/manganese`
warning: running `mn` without the 'install-deps' feature falls back to just doing nothing!

justfile for MnemOS
```

Then when running melpomene:

```
$ cargo mn melpomene
warning: skipping duplicate package `run-wasm` found at `/Users/james/.cargo/git/checkouts/embedded-graphics-web-simulator-ef34bb7f9f200182/0f07b1e/examples/animated/run-wasm`
    Finished release [optimized + debuginfo] target(s) in 0.31s
     Running `target/release/manganese melpomene`
warning: running `mn` without the 'install-deps' feature falls back to just doing nothing!
cargo run --profile release --bin melpomene -- 
warning: skipping duplicate package `run-wasm` found at `/Users/james/.cargo/git/checkouts/embedded-graphics-web-simulator-ef34bb7f9f200182/0f07b1e/examples/animated/run-wasm`
```

hawkw, 2023-09-18:
The "running `mn` without the 'install-deps' feature ..." warning was added by me on purpose because i wanted to make it clear that mn wouldn't actually install stuff without that feature. but, the `cargo mn` alias is _supposed_ to pass that feature, so it seems like that's a bug on my part. Will fix.

The "warning: skipping duplicate package `run-wasm` ..." warnings show up regardless of whether or not you use `mn`; i get them when doing a normal `cargo build` as well. I assume this is due to Pomelo enabling a wasm feature in `embedded-graphics`, but i don't totally get what's going on there...

jamesmunns, 2023-09-19:
Reopening to track the second error, and tagging @spookyvision

spookyvision, 2023-09-20:
the `run-wasm` part is now fixed

commit ba81b098 (2023-09-18): fix(mn): use correct "_any-deps" feature name (#286)

Manganese uses a "private" feature flag called "_any-deps" to determine
whether any bindep dependency feature flag is enabled. Unfortunately,
some parts of the code mistakenly use the name "_any_deps" (with a
second "_" instead of a "-") to refer to this feature, which is
incorrect. This results in Manganese always emitting a warning that no
deps were installed, even when this is not actually the case.
Surprisingly, Rust doesn't warn you when you reference a feature flag
that doesn't actually exist, which means we never caught this issue in [... cut]
closed 2023-09-20 by spookyvision